3DXTech/Katana #6: Instances struggle with Rafts

cdg3d, 2023-11-09:
**Is your feature request related to a problem? Please describe.**
PrusaSlicer (PS) uses Instances where it duplicates GCode rather then reslicing for multiple parts.  While this is normally good as it saves time, when utilizing a raft this gets annoying.  PS will generate rafts where it is printing in areas where it has already printed other rafts, instead of merging them.  
[image]
PS will also auto-arrange based on part size, and will not attempt to correct for raft size.
`Add Part > [+] > [A]`
[image]
[image]

**Describe the solution you'd like**
Still open to ideas.  My best idea right now is to have PS separate instances from raft.  This means the slicer will regenerate the raft each time, but would allow multiple instances to use the same raft rather then each have their own raft.

**Describe how it would work**
With lots of coding magic

**Describe alternatives you've considered**
If that is impossible, I would say that as long as we could take raft size into account when Arranging parts that would be a good backup.  This might be necessary regardless because Arrange will also put the part on the edge of the build area and then build the raft outside the area.
[image]

**Additional context**
Add any other context or paint drawing about the feature request here.

cdg3d, 2023-11-09:
This should NOT delay launch of Katana, as there is a solution:
`Select all Instances > [Right Mouse] > "Set as a Separated Object" > Select all Objects > [Right Mouse] > "Merge"`

ericedson98, 2023-11-09:
@cdg3d When Prusa creates a raft for an object, it's setup so it's completely unaware of all other objects and rafts it may interact with until it's fully sliced and shown in the previewer. I will attempt to come up with some nifty way of handling this, potentially merging the touching objects mid slice?