3DXTech/Katana #7: 1st and Other Layers filament temperatures not working correctly.

jrgGearbox3d, 2023-12-06:
### Description of the bug

When printing a raft, first layer of Model gets set at the very beginning of the G-code, but is then changed to Other layer temperature after the raft is complete, but no first layer of Model has been extruded.

### Project file & How to reproduce

Print any part with a raft.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

N/A

### Operating system

Winblows 11

### Printer model

HT2 Jamin 10047.

cdg3d, 2024-01-05:
This is fairly standard behavior, on cura as well the first layer temperatures only apply to material being laid on the build sheet, we've never been able to utilize first layer temps on model materials.  If we could figure out a way to add this feature, it could be really nice

ericedson98, 2024-02-21:
This feature has been added in commit b9d98c9

closed 2024-02-21 by ericedson98