3DXTech/Katana #13: World Domination Chungus

cdg3d, 2024-02-07:
**Is your feature request related to a problem? Please describe.**
Every time I want to print a Chungus I have to add him from my computer.

**Describe the solution you'd like**
It'd be much simpler if you could easily add him from the shape gallery, he could potentially replace the bunny since he himself is a fat rabbit. 
[image]

**Describe how it would work**
Right click to add shape, select gallery, boom, world domination
https://netorg4911482.sharepoint.com/:u:/r/sites/3DXTECH/Shared%20Documents/ENGINEERING/PROJECTS/00XX%20SLICER/Slicer%20Meshes%20and%20Images/[BigChungus.stl](https://netorg4911482.sharepoint.com/:u:/r/sites/3DXTECH/Shared%20Documents/ENGINEERING/PROJECTS/00XX%20SLICER/Slicer%20Meshes%20and%20Images/BigChungus.stl?csf=1&web=1&e=chL30L)?csf=1&web=1&e=chL30L

**Describe alternatives you've considered**
World Domination must occur, there is no alternative

**Additional context**
We don't need to be too vocal or obvious about this addition, just make it happen.

cdg3d, 2024-02-07:
His image is trademarked by Warner Bros... might not be the wisest decision

closed 2024-02-07 by cdg3d

ericedson98, 2024-02-08:
🤣