3DXTech/Katana #16: Calibration Features

cdg3d, 2024-02-21:
OrcaSlicer has added an entire calibration section to their slicer which is based off of Bambu Studio which is based off of PrusaSlicer.  @Jrauwerter has mentioned that these features are quite useful in dialing in new materials and profiles.  Was curious about the ability to add some of these features to Katana

Wiki for the feature: https://github.com/SoftFever/OrcaSlicer/wiki/Calibration 
I believe the first release where they added this: https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.4.1

ericedson98, 2024-02-22:
@cdg3d Is flow rate the extruder multiplier under filament settings?

cdg3d, 2024-02-22:
Yes it should be

ericedson98, 2024-02-22:
Adding extruder multiplier will be a very big task due to PrusaSlicer being setup to only allow manipulating certain print settings to objects. It doesn't allow filament settings to be manipulated per object. However I think I have found a way to do this without spending weeks refactoring all that code.